umigv/goat #69: Integrating the existing motor control node with move_base node

Jennifer-M, 2017-10-29:
- make sure that the current "motorctrlPWM" program on github compiles and runs with Arduino
- Integrating the existing motor control node with move_base node [in short, make sure they work together well]
- investigating and giving an estimation of the relationship between the power output and robot speed/ turn angle (this might require you to know the weight of the robot, but for now, you can double the weight of the frame as an estimation)[would be ideal if the Arduino signal magnitude can be expressed as a function of the speed and/or turn angle ]

andrewchouman, 2017-10-29:
I'm working on this too

Jennifer-M, 2017-11-18:
Chris Marsh [10:02 PM] 
https://www.dimensionengineering.com/info/arduino

This link contains an Arduino library for sabertooth. We might want to rewrite a portion of our code using this standardized library.

closed 2018-01-31 by atish3