umigv/goat #77: Put all sliders on frame

yomn61, 2017-10-30:


austone, 2017-10-31:
Looks like you may have forgotten to put sliders on the inside longitudinal rails.

arnwill, 2017-11-01:
I really hope not but it is definitely possible. I’d have to see it in person to know for sure but The other possibility is that the rails were not flush and someone moved the frame. We ran into that problem while building it. Since the sliders aren’t connected to anything, if the rails are not flush and the frame is tilted the sliders can sometimes manage to fall out. If any additional sliders need to be added I can take care of it on Sunday… but the sooner we can connect gear boxes and pillow blocks to the sliders the better off we will be.

Sincerely,

Will Arnold

From: austone
Sent: Tuesday, October 31, 2017 6:09 PM
To: umigv/atlas
Cc: Will Arnold; Assign
Subject: Re: [umigv/atlas] Put all sliders on frame (#77)

Looks like you may have forgotten to put sliders on the inside longitudinal rails.

austone, 2017-11-01:
In my experience, it is a bit easier to add sliders to a part then install it than it is to screw into the t-nuts through the part. It'll work out either way

closed 2017-11-06 by austone