umigv/goat #79: Verify that gearboxes can be oriented single file if inverted relative to one another.

austone, 2017-10-31:
If the gearboxes can be mounted at the same point on the length of the vehicle, we can save a lot of internal space. This is only possible if the motors can be mounted vertically flipped relative to one another so that the motors don't contact one another.

yomn61, 2017-11-01:
After placing the gearboxes on the CAD at the far edges of the robot, the timing belt would have to be 70 inches, where the closest sizes for the timing belt are 60 inches and 81.7 inches. So, I have moved the gearboxes in the front to the back of the unhinged section. This way, it is out of the way of any sensors or electrical components that would need to go there. The size of the timing belt is within the vicinity of 60 inches (which has been already ordered).

yomn61, 2017-11-01:
This issue will leave the under review column when it is confirmed that the timing belt will not have an issue with the format proposed.

closed 2017-11-06 by austone