umigv/goat #119: Preliminary Safety and Practices Document

cpmarsh, 2017-11-19:
We need a document with input from the respective coordinators regarding safety and other issues that may arise in the course of their work. The goal of this is to prevent incidents such as the robot falling off a table to happen.

Gregory-Meyer, 2017-11-20:
Any ideas what this might look like for a more software-based pod like Sensors? Aside from not running rm -rf /* I can't really imagine a lot of safety issues with our work.

In any case, I think any time we're doing work in the shop on physical devices - with sensors, the platform, the laptop - someone should know about it at the least. We don't necessarily need to have a buddy system to open the laptop, but there should be stronger rules for more dangerous items.

MishFTW, 2018-02-03:
http://atlas.umigv.com/atlas/safety-guidelines