umigv/goat #135: IMU Dewiggler

Gregory-Meyer, 2018-01-14:
The IMU data output from transform_imu wiggles around. Write an exponential mean filter to take sensor_msgs/Imu input, take the exponential average of the orientation, then output a filtered sensor_msgs/Imu.

Gregory-Meyer, 2018-02-01:
Work abandoned, taking responsibility for this now

kushalj1997, 2018-02-01:
This is resolved - the IMU wiggle was an illusion of oscillation of the gyro data above and below zero. When the IMU is transformed, we see the oscillation of the axis in the Y component of the transformed data also, and I've confirmed this same oscillation appears by showing that the non-transformed data also exhibits this wiggle. Looking at the data side by side, the transformed data corresponds to the raw data from the gyroscope, so there is no issue with the code either.

closed 2018-02-01 by kushalj1997