umigv/goat #174: Set up development NUC

MishFTW, 2018-02-12:
Our current workflow is horseshit and depends a lot on the laptop. Team members rely on using their personal computers and we fall short. 

We need to set up 1 NUC for **production** on the robot that will be readily deployed, and one NUC for development purposes. Environment should be set up either via scripts or continuous integration (maybe later) and simply use Github to pull the latest software.

cpmarsh, 2018-03-18:
Nico has now put SSH capability onto the NUC

MishFTW, 2018-03-18:
http://atlas.umigv.com/atlas/ssh-access

andreseg9726, 2018-04-01:
We downloaded the following on the nuc:
*movebase node and parameter files
*simulation node
*hector plugins for gazebo
*turtlebot teleop

closed 2018-05-06 by ramicaza