umigv/goat #182: Simulate RPLidar on gazebo

andreseg9726, 2018-02-17:


atish3, 2018-02-18:
The descriptions in the comment are based on the tutorial robot we pulled from https://github.com/richardw05/mybot_ws.git.

1. We can adjust the laser range and the angle through which the laser sweeps, as well as the rate of rotation of the laser. The laser was set to a range between 0.1 and 30.0 units, and it was set to sweep through an angle of 180 degrees (-pi/2 to pi/2). We can easily adjust both of these parameters.

2. The camera produces a 3D image. The camera can see objects placed in the field of view. The camera produces an active video of what's happening in the Gazebo simulation. 

3. We're not sure how to output the information gathered by the sensors to the specified nodes.

andreseg9726, 2018-02-19:
Good job figuring out that you can easily adjust the range and sweep angle of the laser. If I'm not wrong, the RPLidar does 360 degree turns so you can try setting it to that (you can confirm with the sensors team). 

As for communications, you can tap into the topic of the laser with the command rostopic http://wiki.ros.org/rostopic and see what it's outputting. If this output changes when you put something in front of the robot then it's working. 

Don't worry about integration with other nodes yet, that will be on another ticket once you're done with this.

closed 2018-03-25 by andreseg9726