umigv/goat #199: Sensors Housing

yomn61, 2018-03-10:
Details to come

caramast, 2018-03-11:
What's the price constraint on this?

MishFTW, 2018-03-16:
Let's keep it under $300 if possible

yomn61, 2018-03-16:
https://umigv.slack.com/files/U7389QDMY/F9N6LRUUD/enclosure_system

yomn61, 2018-03-18:
Is there a layout of how the components will fit inside the box if we go with first option?

caramast, 2018-03-18:
I haven't made one yet

yomn61, 2018-03-18:
Can that be made by Tuesday so we can confirm how mounting will happen and that it feasibly can happen with the space in the box?

caramast, 2018-03-18:
What platform should it be on and where could I find a list and dimensions of each component that is going to be in the enclosure

yomn61, 2018-03-19:
the components all should be found in the top level directory on Fusion. (Power supply for the Jetson, Jetson, NUC) I will systematically try to add the other parts into Fusion in the next hour or so.

yomn61, 2018-03-29:
Sensor Housing has been bought, will be here Thursday or Friday

yomn61, 2018-03-29:
It has been bought and is en route.