umigv/goat #200: Paneling Planning and CAD

yomn61, 2018-03-10:
Details to come

camkab, 2018-03-20:
Began to document panel requirements and sizes along with current stock in google doc in spec 1>platform>design ideas: https://docs.google.com/document/d/1PozL3-6XlxD36BcW95d-KjuO-Ly-15062PtyeezVdUA/edit

camkab, 2018-03-21:
Top plate and bottom plate will be laser cut out of HDPE on Saturday (HDPE picked up on Friday, ordered on 3/21) according to Cameron's coordination with the larger laser cutter.
Power shelf will be laser cut at the same time as the top and bottom plates.
The plates will be cut to shape, but not necessarily have holes for standoffs-those can be drilled by hand as needed.
The sides and front/back will be handled later (will not hold back teleop requirement by 3/28), likely in the Dude laser cutter.

yomn61, 2018-03-22:
The Grainger order has been placed, will notify the team about when it is ready

camkab, 2018-03-28:
The Document is still being polished. Will need input for deciding material, thickness, and potential decals/designs/sponsor logos. Will put this ticket into review and then close, as the above issues will need to be decided for the manufacturing ticket.

camkab, 2018-03-29:
Design reviewed with Madhav and Ramiro, no notable changes. Agreed to use 1/8" thickness material. Material choice will be discussed at the Leads Meeting by Madhav, along with decals and other logos. This ticket will be closed now as these issues will be resolved in the manufacturing ticket.

closed 2018-03-29 by camkab

camkab, 2018-03-29:
Document has been finished: 
https://docs.google.com/document/d/16keXypY_W7WG-2f76Sl7KG-snRSxHmYssUAf9YEHjD8/edit