umigv/goat #214: Generate Encoder Covariance Matrix

Gregory-Meyer, 2018-03-11:
In order for our state estimation nodes to work properly, we need covariances on every input. Since the encoder node is written by us, we need to find its covariance matrix. First, we need to find out how to generate the covariance matrix, then we need to generate, and then we need to incorporate it.

MishFTW, 2018-03-25:
Linear motion complete, angles incomplete;

MishFTW, 2018-03-25:
Kaizen:
Joseph doesn't know how to use robot on his own (run ROS); could have reached out to Greg/Sam

We may be missing some communal information on GOAT operating procedure; missing documentation on robot startup procedures @Gregory-Meyer

Gregory-Meyer, 2018-03-25:
Added SLAM startup procedure to [Atlas](http://atlas.umigv.com/atlas/docs).

MishFTW, 2018-05-03:
Need to recalibrate

Gregory-Meyer, 2018-05-13:
The requirements for this have changed since encoder_odometry has been modified to output TwistWithCovarianceStamped messages. Need to test that the velocity output matches expected values for various speeds.

closed 2018-11-18 by cedricIGV