umigv/goat #218: Atlas - Platform ENGR Documentation

MishFTW, 2018-03-14:
Update the http://atlas.umigv.com/atlas/platform with the following:
* **System Requirements:** Describes what the system must do in terms of function and performance. Maps system requirements to user needs. Forms the basis for the design/build phases.
* **System Acceptance & Test Criteria**: Describes steps/information required for testing functional accuracy and robustness of sub-systems and overall system. Iteratively updated as development proceeds.
* **System Diagram**: A visual representation of all the components of the system and how they interact with other systems. This also includes a bill of materials on the final build with appropriate properties such as cost, quantity, reference to CAD, etc.

yomn61, 2018-03-25:
Will be working on this on Sunday, 3/25, meeting

yomn61, 2018-04-10:
pushed back due to low priority

closed 2018-05-24 by MishFTW