umigv/goat #242: Integrate Status Light

cpmarsh, 2018-03-25:
When the status light is on, it must be implemented to reflect the current status of the robot (enabled, autonomous, etc) as according to the rules. Currently, the most efficient way to do this would probably be to modify the outputs of the motor arduino, and pass additional information along over ros_serial. The amount of work for this task will depend on how much the ROS nodes have to be modified.

MishFTW, 2018-05-24:
Thinking of connecting to Arduino Mega for actual control of RGB strip

Need a node to spit out a boolean value for autonomy which the mega will listen to and determine status light logic to meet design requirement of flashing for autonomy and solid for powered

gtheller, 2018-05-28:
I haven't seen the status lights before or tried to program RGB leds. If I'm still taking point on this, I'll need someone from controls to walk me through what the different statuses need to look like.

closed 2018-06-08 by MishFTW