umigv/goat #275: Run a Heat Test on the Jetson and NUC

yomn61, 2018-04-04:
Find out how hot the Jetson and the NUC get under nominal conditions inside the Sensors Box.

samjhall, 2018-04-08:
I ran stress-ng on the NUC for about 20 minutes with the NUC almost entirely contained within its box and it got hot, but not close to overheating. Considering the fact that the compute box will be far more open, the competition runs cannot be more than 6 minutes, and the box can be vented between runs, I believe the NUC's heat wont be a problem. I also tested the Jetson GPU with GLMark and the heatsink didn't even get warm.

Gregory-Meyer, 2018-04-08:
Given the short duration of runs, installing a fan was not determined to be necessary

closed 2018-04-08 by Gregory-Meyer