umigv/goat #339: Design and 3d print battery brackets

camkab, 2018-10-04:
Design brackets that will securely hold the battery in its new position while still allowing accessibility to pull the battery out. Mind where the bolts will hold the bracket so they don't interfere with support bars underneath.

3d print the brackets.

camkab, 2018-10-04:
First set of printed brackets had few issues:
Printer failed and caused offset mid print, needing to reprint the brackets.
The key did not fit into the keyhole in the bracket, requiring a better tolerance to account for the printer's printing error.

camkab, 2018-10-04:
All pieces should be done printing by now and ready to be picked up and installed.

camkab, 2018-10-07:
Some changes to be made for future models, sticky notes defining the changes were handed to the owners of this task.

karolinarak, 2018-10-07:
Changes were made on Autodesk files from the definitions on the sticky notes, brackets were printed as well

closed 2018-10-18 by camkab