umigv/goat #488: ROS training development

trsabb, 2020-07-02:
Prepare a set of presentations, documents, and/or demos to give new members the basic understanding of ROS needed to contribute to ARV. Work off of last year's trainings [here](https://docs.google.com/document/d/1m1qNfrz0j9Rx-6F4kuEiecTMGefEPyAE-XDxsPhmBks/edit) and [here](https://docs.google.com/document/d/1sVrUhGcnEE8AiOtJM_l5Piec_tFZ29mryQfrYteksiM/edit) if needed. This training should ideally be 6 - 7 hours long (roughly two meetings) and assume no prior experience, but a willingness to learn.

hvak, 2020-07-15:
https://docs.google.com/document/d/1m4KMcZlsYTl6gUXYzT16zWDP7sM9kuJZorszk932WrI/edit?usp=sharing

Elton-Lin, 2020-07-26:
@hvak Looks pretty good! Is there anything you'd like me to add? Just let me know. Also, I can try finding some visual examples about topics and nodes.

Elton-Lin, 2020-07-26:
This is a pretty good [video](https://www.youtube.com/watch?v=0BxVPCInS3M&list=PLE-BQwvVGf8HOvwXPgtDfWoxd4Cc6ghiP) series for ROS intro. It's a bit lengthy, but I think we can link it somewhere in the doc as an extra resouce.

hvak, 2020-07-26:
I think some visual examples/diagrams would be a great addition if you can find or create any. Also, adding additional resources is a good idea

hvak, 2020-08-19:
Slide format with publisher/subscriber example [https://docs.google.com/presentation/d/1DWYZkS_Sxfgd9XESKr7-G1oshUhvzolYU-qc6s29XPI/edit?usp=sharing](https://docs.google.com/presentation/d/1DWYZkS_Sxfgd9XESKr7-G1oshUhvzolYU-qc6s29XPI/edit?usp=sharing)